abetusk/gbr2ngc #3: Implement Aperture Macros (AM)

abetusk, 2018-02-07:
Aperture macros aren't implemented, implement them.  There's a complication in that the aperture macro can reference aperture definitions and also have "arithmetic expressions with other variables", so we're starting to need to implement a real language.

The first step is to make sure the aperture macros can parse lines that are simple, without references to other aperture definitions or contain arithmetic operations. The next step is to add aperture definitions.  Finally, the last step is to add arithmetic operations.

abetusk, 2018-02-20:
Aperture Definitions (AD) can reference Aperture Macros.  Instantiating an AD from an AM only needs to evaluate the AM once at load and the geometry can be referenced.

I'm thinking of using a stripped down version of [tinyexpr](https://github.com/codeplea/tinyexpr) for the calculator functions.

The AD function can mostly be kept except for an alternate code path to reference an AM if used.

abetusk, 2019-02-28:
`tinyexpr` has been added to `gbl2ngc` along with code to evaluate aperture macros in the branch `gerber-extensions`.

Briefly, aperture macros are handled by extending the `gerber_state_t` structure in `gerber_interpreter.h`. A linked list of newly defined `am_ll_node_t` structure is used to create a linked list of aperture macros, where each node holds a string array for each aperture macro command.

The parameters to each aperture macro are shoved into a `tinyexpr` context as global variables whose names are the `$<param_position>`. In each line of the evaluated aperture macro, lines are parsed either creating aperture definitions and evaluating each `AD` parameter with a call to the `tinyexpr` engine or are used to create a new variable. In a "new variable" definition, the right hand side of the equal is evaluated with a `tinyexpr` context that has all previous variable names for this aperture macro set up in it's environment.

`gerber-extensions` need to implement the rest of the Gerber spec as well as provide some tests before it's merged into release.

abetusk, 2019-05-02:
resolved by 2e149e48359e55763310077dbe761446792523f1

closed 2019-05-02 by abetusk

XRay3D, 2019-12-06:
https://github.com/XRay3D/GERBER_X2