abetusk/gbr2ngc #7: Implement block aperture (AB)

abetusk, 2018-02-07:
Section `4.6` of rev. 2017.11 of `The Gerber Format Specification`.

Block apertures can be nested and reference other aperture definitions.

example:
```
G04 Ucamco copyright*
%TF.GenerationSoftware,Ucamco,UcamX,2016.04-160425*%
%TF.CreationDate,2016-04-25T00:00;00+01:00*%
%TF.Part,Other,Testfile*%
%FSLAX46Y46*%
%MOMM*%
G04 Define standard apertures*
%ADD10C,7.500000*%
%ADD11C,15*%
%ADD12R,20X10*%
%ADD13R,10X20*%
G04 Define block aperture D100, consisting of two draws and a round dot*
%ABD100*%
D10*
X65532000Y17605375D02*
Y65865375D01*
X-3556000D01*
D11*
X-3556000Y17605375D03*
%AB*%
G04 Define block aperture D102, consisting of 2x3 flashes of D101 and 1
flash of D12*
%ABD102*%
G04 Define nested block aperture D101, consisting of 2x2 flashes of D100*
%ABD101*%
D100*
X0YD03*
X0Y70000000D03*
X100000000Y0D03*
X100000000Y70000000D03*
%AB*%
D101*
X0Y0D03*
X0Y160000000D03*
X0Y320000000D03*
X230000000Y0D03*
X230000000Y160000000D03*
X230000000Y320000000D03*
D12*
X19500000Y-10000000D03*
%AB*%
G04 Flash D13 twice outside of blocks*
D13*
X-30000000Y10000000D03*
Copyright Ucamco NV
Having a question or remark about the spec? Please contact us at gerber@ucamco.com
83X143000000Y-30000000D03*
G04 Flash block D102 3x2 times*
D102*
X0Y0D03*
X0Y520000000D03*
X500000000Y0D03*
X500000000Y520000000D03*
X1000000000Y0D03*
X1000000000Y520000000D03*
M02*
```

output:
[image]

abetusk, 2019-02-28:
This is still a work in progress but the branch `gerber-extensions` should now support Aperture Blocks.

Before that branch gets merged into release I'd like to add the other Gerber extensions (I believe the `SR` and some other `LP` codes are the major ones yet to be implemented) as well as provide some tests.

closed 2019-05-02 by abetusk

XRay3D, 2019-12-06:
https://github.com/XRay3D/GERBER_X2/blob/master/gerber/gbraperture.h
class ApBlock : public AbstractAperture, public QList<GraphicObject> 

https://github.com/XRay3D/GERBER_X2/blob/master/gerber/gbrparser.cpp
void Parser::addFlash()
void Parser::addPath()

abetusk, 2019-12-06:
@XRay3D, Why are you leaving this comment?

commit 00ef99e1 (2019-05-02): resolves #7, #6, #3, #2

* lightly tested
* still need to run through all test files to make sure things look good.
  Assuming they do, I think it's good to merge
commit a584a6dd (2019-05-02): resolves #7, #6, #3 and #2