abetusk/gbr2ngc #8: Allow for file attributes to inform GCode output?

abetusk, 2018-02-07:
With file attributes, there's a facility to communicate information to the GCode output.  One could theoretically communicate bit size, speed, etc.  This might need to be done at a higher level or maybe it's not appropriate to store int he file attributes but it's a possibility.

DariusMontez, 2018-06-06:
Clarification: would the file attributes be attached to the Gerber file or the GCode file? 
I am assuming the user would attach attrs to the Gerber, which gbl2ngc would read and apply before exporting the GCode. 

I don't have much experience with file attrs, but I think I get the gist of it. For example, lets say a Gerber file had an extended file attribute with the text:

```ini
radius = 0
feed = 15
```

Each option matches with one of gbl2ngc's command-line arguments. When gbl2ngc does it's work, it will produce a GCode file containing the line `F15` somewhere at the top, respecting the data in the file attrs. Is this the use case you are describing? Please correct me if I misunderstood :)

My mild research seems to show that file attrs are dependent on the filesystem format, so we might run into issues with portability on Windows/Mac/Linux. We run the risk of the file attrs silently disappearing if the file is copied but the attrs are not preserved. 

Another option would be to put the metadata in the Gerber file itself as a comment.

```gcode
G04 config: zcut=0.004, seek=200 radius=0.02
```

What I don't like about these methods is that the user still has to edit the input file, either by adding attrs or comments. After making a few boards, users will probably settle on some config options that reliably produce desirable results with their particular machine setup. I made PR #12 for config file support so that users can persist those settings. If the user has two types of PCB material that they cut, say 1.6mm and 0.8mm, then they could make one config file for each type. Does that make sense?

abetusk, 2018-06-06:
I was thinking of having the attributes in the Gerber side of things.  The Gerber specification has an "Attributes" definition for user defined attributes.  If you look in section 5 of the [Gerber specification](https://www.ucamco.com/files/downloads/file/81/the_gerber_file_format_specification.pdf) there is a property called `File Attributes (TF)` which can be used to define user defined file attributes.

I was thinking this could be used to define the different radius, feeds, etc. that one might need.

I don't know what's common practice so I don't know how reasonable it is to expect those attributes to carry over when the Gerber file is being shuttled around. Since it's in the specification, I would assume that it's fine but I don't know.

In terms of motivation, I don't think putting options in the `File Attributes` field in the Gerber file is mutually exclusive to having a config file. I would lean towards using config files for these types of values but the config file is tied t the tool whereas the field attributes are tied to the file, so I can see it both ways.

I saw your pull request but haven't had time to take a look. I'm hoping to get to it by no later than end of weekend.

And thank you for the interest and pull request!

DariusMontez, 2018-06-06:
Ah! I did not realize you were talking about Gerber File Attributes and not filesystem attributes -- how confusing :). Yes that makes more sense now and I think you idea is easily feasible and useful.