abetusk/gbr2ngc #10: Allow for extra GCode parameters

abetusk, 2018-02-25:
Many other tools allow some level of control over the output GCode.  These should be added.

I imagine a good start is:

* **Header:** to appear at the beginning of the GCode file
* **Footer:** to appear at the end of the GCode file
* **Rapid motion:** to appear after or with every rapid motion (`G01`)
* **Cut motion:** to appear with every cut motion (`G00`)

Each of these options should take in either a string at the command line or a file option.  They each should have reasonable defaults.

A first iteration can have raw strings/files to be added.  It might be interesting to either have a format string option or a small DSL to allow for more in depth GCode manipulation.

DariusMontez, 2018-06-06:
I have limited CNC experience, and I understand the value of header and footer blocks. but can you explain a use case for rapid/cut blocks? Would you be adding params inline, like `G1 X3 Y-5 MY EXTRA CODE` or in separate line, like

```gcode
G0 X3 Y-5
MY EXTRA CODE
```

DariusMontez, 2019-02-09:
Can we safely say that PR #13 closes this issue, or should we also address extra blocks for G0 and G1?

abetusk, 2019-02-09:
Sounds good, closing.

closed 2019-02-09 by abetusk