abetusk/gbr2ngc #18: Conversion fails at 274X offset command

TJNII, 2020-03-19:
Getting a parse error when attempting to convert a RS274X file.  I can successfully open the file using gerbv.

```
$ gbr2ngc /host/dtmf_surf/protoboard.cmp -o /tmp/protoboard.cmp.nc
PARSE ERROR: bad SR format, expected character 'X' at line 3 '%OFA0B0*%'

$ head /host/dtmf_surf/protoboard.cmp 
G75*
G70*
%OFA0B0*%
%FSLAX24Y24*%
%IPPOS*%
%LPD*%
%AMOC8*
5,1,8,0,0,1.08239X$1,22.5*
%
%ADD10C,0.0000*%
```

`%OFA0B0*%` appears to be a standard command per https://d1.amobbs.com/bbs_upload782111/files_11/ourdev_450330.pdf

abetusk, 2020-03-20:
Thanks for the bug report!

That standard looks to be from 2001. The standard that I used as a reference was from 2018 and has the `OF` (Offset) command listed as one of it's deprecated format commands. See section 7.1.8 of, as of this writing, the most recent [revision 2019.06 of The Gerber Format Specification](https://www.ucamco.com/files/downloads/file/81/The_Gerber_File_Format_specification.pdf?dd1347f8978ee2fb4532ef5613d) from the Ucamco website.

I'm not opposed to implementing deprecated features if it's really wanted but it seems like it might be bad form. I think a better way to go would be to offload this task to another tool that brings older Gerber files up to date with the latest standard.

Was the file you were trying to convert old? Would you be satisfied with a tool that converted older Gerber files to remove the deprecated functions? Do you know of any Gerber manipulation tools?

It'd be nice to have at least some suggestions in place so if anyone ran into the same problem in the future, they would have options instead of a blanket "we don't do that".

TJNII, 2020-03-20:
Oh okay, I did not expect the commands to have changed.  I am using an old CAD program, if the file formats have changed in the last few years that explains some of the frustration I was having yesterday.  I'll chalk this up to user error and look into upgrading my CAD program, then.  (It was on the to-do list anyway.)

closed 2020-03-20 by TJNII