abetusk/gbr2ngc #22: Curved paths are missing from silkscreen

jes, 2021-02-16:
I'm not sure precisely what the problem is here, but I have observed that the G-code for my silkscreens is missing the curved paths.

I've attached a (zipped, to appease github) Gerber file that causes the problem, along with a screenshot of the G-code path as rendered by FreeCAD, and the PCB as rendered by KiCad.

Curiously, the text is fine. It is only the nub on the graphic that is a problem. I wonder if the text is composed of line segments and the nub is a curved path?

[nubgerb.zip](https://github.com/abetusk/gbr2ngc/files/5990995/nubgerb.zip)
[image]
[image]

jes, 2021-02-16:
Line 29 of the gerber file appears to be a curved path indeed:

`G02X126254000Y-72838000I-1000000J0D01*`

This looks very similar to the actual G-code required to form a curved path, so it may be quite an easy fix.

abetusk, 2021-02-17:
Oof, well, this is embarrassing. Apparently I hadn't implemented `G02` and `G03` for straight line segments.

The parsing code is there and, if I remember correctly, I did the interpolation for regions, I just never got around to adding it in for the linear segments for some reason.

At any rate, it looks to be a quick fix as most of the interpolation etc. is implemented, so it's just about parsing the corresponding `G02` and `G03` lines correctly.

FYI, the engine for the polygon unions is Angus Johnson's [clipperlib](http://www.angusj.com/delphi/clipper.php) which acts on linearlized polygons, so all curves get interpolated to linear segments which are then fed into clipperlib to do polygon unionization.

I've included a picture of what the output is now. I've also added the `G03` functionality. This is minimally tested so please keep on the look out for any anomalous behavior.

[image]

Thanks for the test Gerber, this greatly speeds up debugging!

Would you mind if I included the Gerber file you provided in the examples or tests?

closed 2021-02-17 by abetusk

jes, 2021-02-17:
Thanks, that was fast! Clipperlib sooks cool. The screenshot looks perfect. And yes you are welcome to use the Gerber file. I created it by creating a blank PCB in KiCad and adding a footprint for a DIP-4 socket or something like that.

commit 4aaaa504 (2021-02-17): resolves #22

* g02 and g03 straight line segments weren't added? fixed now, hopefully