abetusk/gbr2ngc #25: Bug, and then not generate the output file

alphaonex86, 2023-06-04:
Hi,
it bug with error:
ERROR: jump pos MUST be > 0, is -1, exiting (start 2, n 155)
ERROR: jump pos MUST be > 0, is -1, exiting (start 2, n 155)
Command:
gbr2ngc --input nuco-v-F_SilkS.gto --radius 0.0025 --output nuco-v-F_SilkS.ngc
[nuco-v-F_SilkS.gto.gz](https://github.com/abetusk/gbr2ngc/files/11647511/nuco-v-F_SilkS.gto.gz)

abetusk, 2023-06-06:
Thanks for the file. `gbr2ngc` has many, many problems with it.

I hope to have a closer look when I get more time.

abetusk, 2023-12-27:
This is mostly a note to myself:

I believe this is an issue with how I calculate the regions. I made some assumptions about where the start and end point of the region were that look to be violated in the provided Gerber file.

There are some circle regions that intersect rounded rectangle regions that look to share some points, as in the following:

| | |
|---|---|
| [image] | [image] |

For comparison, here is the relevant region in gerbv:

[image]

I believe those circles correspond to the circular regions with rounded rectangles.

This could also be an issue with the Gerber parser not correctly picking up the relevant regions. Regardless, this is clearly a bug in gbr2ngc.